oyvindln/vhs-decode #4: TBC issue on tape with ghosting

oyvindln, 2020-02-26:
Found by zcooger This tape has some issues causing the TBC to mis-detect the hsync starts:

https://cdn.discordapp.com/attachments/654077758539825152/682294635502239786/frame_pal_source_4367.png

oyvindln, 2021-12-19:
Currently the sync detection only checks for the start of hsync, it may be better to use the end of hsync as that should be less affected by overshoot etc.

joshenders, 2025-09-27:
> Found by zcooger This tape has some issues causing the TBC to mis-detect the hsync starts:
> 
> https://cdn.discordapp.com/attachments/654077758539825152/682294635502239786/frame_pal_source_4367.png

Looks like discord attachments are no longer available due to this: https://www.bleepingcomputer.com/news/security/discord-will-switch-to-temporary-file-links-to-block-malware-delivery/ 

Found a copy of `frame_pal_source_4367.png` on wayback machine and posting here for future reference: [image]