oyvindln/vhs-decode #8: Crash samples

Zcooger, 2020-12-14:
Following samples crash vhs-decode with the error:
```
file frame X unknown
Warning! Chroma signal clipping.
ERROR:root:Unable to find any sync pulses, jumping one second
ERROR - please paste the following into a bug report:
current sample:  X
arguments:  Namespace(cxadc=False, cxadc3=True, dod_threshold_a=None, dod_threshold_p=None, dodod=True, infile='/mnt/SAMSUNG 1000/Users/Zcooger/Desktop/capture.wav', length=1000, level_adjust=0.2, ntsc=False, ntscj=False, outfile='capture', pal=True, seek=-1, start=0, threads=16, track_phase=None)
Exception:  'FieldPALVHS' object has no attribute 'burstmedian'  Traceback:
  File "/home/zcooger/Desktop/vhs-decode-doc/vhs-decode", line 215, in <module>
    f = vhsd.readfield()
  File "/home/zcooger/Desktop/vhs-decode-doc/lddecode/core.py", line 2941, in readfield
    fi, needFiller = self.buildmetadata(f)
  File "/home/zcooger/Desktop/vhs-decode-doc/vhsdecode/process.py", line 844, in buildmetadata
    if math.isnan(f.burstmedian):
```
[vhs-decode-crash-samples-14.12.2020.7z.001.txt](https://github.com/oyvindln/ld-decode/files/5686837/vhs-decode-crash-samples-14.12.2020.7z.001.txt)
[vhs-decode-crash-samples-14.12.2020.7z.002.txt](https://github.com/oyvindln/ld-decode/files/5686840/vhs-decode-crash-samples-14.12.2020.7z.002.txt)
[vhs-decode-crash-samples-14.12.2020.7z.003.txt](https://github.com/oyvindln/ld-decode/files/5686841/vhs-decode-crash-samples-14.12.2020.7z.003.txt)
[vhs-decode-crash-samples-14.12.2020.7z.004.txt](https://github.com/oyvindln/ld-decode/files/5686843/vhs-decode-crash-samples-14.12.2020.7z.004.txt)

Zcooger, 2020-12-14:
[image]
[image]

oyvindln, 2020-12-14:
Will look into it when I have time, probably some decoding failure ended with the variable not being defined.

VideoMem, 2021-05-16:
Both of these sample seems to decode without issues on the current vhs_decode branch.

closed 2021-05-21 by oyvindln