oyvindln/vhs-decode #18: TV System support

oyvindln, 2021-04-19:
VHS and U-Matic (vhs-decode part):
- [x] NTSC
- [x] PAL (normal 625 line)
- [ ] SECAM-L
- [ ] MESECAM (VHS specific)
- [ ] PAL-M
- [ ] PAL-N (Not sure if there is any practical difference from normal PAL on tape)

CVBS and ld-chroma-decoder part:
- [x] NTSC
- [x] PAL (normal 4.43 625 line)
- [ ] PAL-60
- [ ] PAL-M
- [ ] PAL-N
- [ ] SECAM
- [ ] NTSC 4.43

cher-nov, 2026-09-01:
Just FYI: there's a similar project supporting SECAM decoding from VHS.
https://github.com/grishka/AVDecoder - _partially in Russian, use online translator_

harrypm, 2026-09-01:
> Just FYI: there's a similar project supporting SECAM decoding from VHS.
> https://github.com/grishka/AVDecoder - _partially in Russian, use online translator_

Yup he's in the community, there's still a little bit of work to be done still for SECAM but it's now in the tools, next thing will be some metadata updates for it and dedicated profiles for the more professional formats as more samples come in now Etinne is all settled in with with the MISRC V2.5.