oyvindln/vhs-decode #20: Tape format support

oyvindln, 2021-04-19:
VHS
- [x] SP
- [ ] 625-line (PAL) LP (lower speeds work-ish but use SP settings)
- [ ] 625 EP/SLP
- [ ] 525-line (NTSC) LP
- [ ] 525 EP/SLP
- [ ] 525 VP (super slow speed on certain panasonics)
- [ ] SVHS and SVHS-ET (nothing yet, but will work-ish with modified rf filters)

U-Matic
-[x] PAL "regular band"
-[ ] PAL hi-band
-[ ] PAL SP
-[x] NTSC "regular band"
-[ ] NTSC hi-band
-[ ] NTSC SP
-[ ] SECAM (does it exist at all?)

Other formats:
- [ ] Video8/Hi8/Digital8
- [ ] Betamax
- [ ] Video2000
- [ ] Philips VCR
- ...etc

etienne-ina, 2023-02-17:
Just a little comment:
Yes, there is a SECAM U-matic, and a SECAM Betamax, and a SECAM VHS. Fortunately, no SECAM V8/Hi8.

Thanks for the outstanding work. Getting pretty good results with CXADC capture of PAL Umatic here (the drop-out compensator is impressive).

If useful, I could provide samples of CXADC captures from low-band U-matic (PAL and SECAM), high-band U-matic / BVU (PAL and SECAM), VHS (SECAM), Betacam (PAL) and maye even SECAM Betamax, Video2000, VCR, and why not 1/2 inch open-reel formats (pre-EIAJ and EIAJ).

oyvindln, 2023-02-17:
Awesome. PAL (low-band) U-MATIC could probably use some filter tweaking still, seemed a bit off from what I remember but helpful with more samples.

And samples of all those formats would be awesome.

etienne-ina, 2023-02-17:
OK nice, will do.
Should I be concerned about the CXADC device being too limited for higher bandwidth formats (such as high-band U-Matic)?

harrypm, 2023-02-17:
@etienne-ina 28msps 8-bit can cover most tape formats but you can upgrade it to a 40msps device with a [crystal replacement](https://github.com/oyvindln/vhs-decode/wiki/CX-Cards) there are lots of little things to improve the abilities of these cards.

etienne-ina, 2023-02-17:
@harrypm Thanks. For now, high-band U-Matic through vhs-decode is giving a picture, it's completely off but it's something.
[image]

oyvindln, 2023-02-17:
Neat, that would be expected as it uses higher frequency carriers for luma (and I think chroma too but don't remember) than "regular band" so need to add a setting for it. I have no idea if 28 msps cxadc will be sufficient to get the higher res variants in good quality but it should at least give something to work with.

etienne-ina, 2023-02-18:
I added U-Matic samples to the GDrive, in two folders according to the VTR used for playback. There are:

- 4 PAL U-Matic samples (including one same tape played on 2 different machines)
- 3 PAL BVU samples
- 1 SECAM U-Matic sample

note : "BVU" (as in Broadcast Video U-Matic) is short for High-Band U-Matic.

Don't hesitate if you have prioritary requests for formats / standards you'd like to have more samples from. It will take a bit of time to do all the formats mentioned before (and probably a capture hardware upgrade).

oyvindln, 2023-10-07:
We need a better table for this  rather than just a checkmark, e.g many formats have a degree of support in the sense that they decode but don't really have correct filtering.

harrypm, 2023-10-07:
Tell me how many fields need adding and I'll just mark down a table for it. @oyvindln