oyvindln/vhs-decode #23: cvbs-decode Chroma decoding struggles on unstable sources like VHS playback

oyvindln, 2021-04-19:
On cvbs samples from unstable video sources like a VHS VCR, the chroma decoding has issues a lot of the time. Additinally there needs to be some sort of agc on the chroma signal based on burst to avoid fluctuations. This isn't an issue on stable sources.

FlashlightET, 2024-03-20:
[image]
[image]
i can confirm this - the left side starts out fine but as it gets to the right side it loses phase

harrypm, 2024-03-20:
@FlashlightET This is why we have vhs-decode 😉 cvbs-decode does work well for live sources and ones that have gone though some level of signal stablisation.

But the SNR on that looks really low, what did you use to capture?

FlashlightET, 2024-03-29:
don't worry i will be getting the setup for vhs-decode in the near future i was just seeing what i could get out of cvbs decode for now

the SNR is fine on the OSD capture when the tape is not playing, and this was a noisy off-air recording which would explain most of the low SNR