oyvindln/vhs-decode #41: JSON file is not written sometimes

itewreed, 2021-12-16:
There seem to have been a change in the logic how the JSON file is written. It is not written anymore after every 250 frames. Instead it is written to the filesystem at the end of the decode process. (Current release as of December 11th)
As this has advantages, there is also the disadvantage, that it will not be written if something in the decode process breaks in-between. This way also a live check of decoding in ld-analyse is not possible anymore.
I have some captures where this happens. The output then is as follows.

assumed_phase: 0
File Frame 5409: CAV Frame #None Unknown                                        
lastline < proclines , skipping a tiny bit
Unable to determine start of field - dropping field
Unable to determine start of field - dropping field
Unable to determine start of field - dropping field
Unable to determine start of field - dropping field
Unable to determine start of field - dropping field
Unable to determine start of field - dropping field
Unable to determine start of field - dropping field
Unable to determine start of field - dropping field
saving JSON and exiting
Cannot build json: 'NoneType' object is not subscriptable
Traceback (most recent call last):
  File "/usr/lib/python3.8/multiprocessing/queues.py", line 245, in _feed
    send_bytes(obj)
  File "/usr/lib/python3.8/multiprocessing/connection.py", line 200, in send_bytes
    self._send_bytes(m[offset:offset + size])
  File "/usr/lib/python3.8/multiprocessing/connection.py", line 411, in _send_bytes
    self._send(header + buf)
  File "/usr/lib/python3.8/multiprocessing/connection.py", line 368, in _send
    n = write(self._handle, buf)
BrokenPipeError: [Errno 32] Broken pipe

The log and tbc files are fine, but can neither be analysed nor used for generating a video. A sample of that behaviour tape10_02_sample.r8 can be found here https://drive.google.com/drive/folders/1OJXtm2IJcVZc3Q7OC146t5ThWyPPJpEL

oyvindln, 2021-12-16:
Hm, did it change very recently? The only thing I've done in that part recently is to move some code into a function but I wouldn't have thought that would have made a difference. This is an issue though yeah, need to see if the json writing can be improved a bit. Additionally right now it rewrites the whole json every time it does update instead of adding to the end which isn't really ideal.

itewreed, 2021-12-16:
It is something, that changed between a commit on October 18th and now. Can also be, that it is something, that was changed by dd86. As I've seen, that you also merged some work from the original lddecode repository in November.

itewreed, 2021-12-16:
Json file writing works till commit 0af1239ddda8f3c058e5d71666480d64c666aa4b , before the gui work and separation of vhs-decode into a function. The NoneType error already happens at the end of the decode in the sample file.

oyvindln, 2021-12-17:
Right, then it's probably something to do with crash/exception handling changing due to it being moved into a function.

commit 865442c5 (2021-12-17): Move json writing to correct indentation

fixes #41
closed 2021-12-17 by oyvindln