oyvindln/vhs-decode #44: Audio & Timecode Alinment For Muxing

harrypm, 2022-01-31:
As noted with a [conversation with NeXT ](https://discord.com/channels/665557267189334046/665834485975351307/937551468163133450) 

We need a timestamps file on TBC generation to fix potential audio sync issues.

tvasenin, 2022-04-20:
There are few reasons why do we need per-field timestamps:

- VCRs generally don't output exactly 50 or 60000/1001 fields per second, meaning audio/video track will be out of sync. In most cases, we want to align audio to video with constant frame rate, hence need the timecodes for global correction.
- In case of any continuity issues (tape, decoding, etc.) it's useful to examine the timestamps for local correction.

harrypm, 2024-01-10:
@tvasenin Since we how have [Auto Audio Align](https://gitlab.com/wolfre/vhs-decode-auto-audio-align) this issue is now more a matter of extra metadata from HiFi-Decode as mentioned in the improvemnet ticket [#77](https://github.com/oyvindln/vhs-decode/issues/77).

closed 2024-01-10 by harrypm