oyvindln/vhs-decode #55: ld-decode.exe not found in Windows

TheGitMpeg, 2022-08-29:
Hello, I want to convert a video into .TBC file extension in NTSC mode, but in my OS, I don't see the "ld-decode.exe" executable file.
[image]
I tried with each one of 3 available folders, but that executable wasn't here.
[image]
Where is that executable file?

harrypm, 2022-08-29:
> 

You are confusing the windows version of the ''LD-Tools'' suite with VHS/LD decode.

As they are python scripts, currently, Ubuntu under WSL 2 is how the majority of windows users use the software suite on dedicated windows systems due to dependencies, having LD-Analyse on windows allows for easy previewing of decodes etc. 

Note the decoder is only for RF data, not ''video'' files for clarity.

oyvindln, 2022-08-29:
I have managed to make slf-contained builds of ld-decode executable for windows with pyinstaller, can update and upload it if there is interest. Don't have one that has all of ld-decode, vhs-decode and cvbs-decode binaris as multiple binaries in one was a bit more tricky.

TheGitMpeg, 2022-08-30:
How to have the ld-decode executable ion Windows 10? "harrypm", you're right.

harrypm, 2022-10-01:
> How to have the ld-decode executable in Windows 10? What commands do I need? "harrypm", you're right.

https://github.com/oyvindln/vhs-decode/wiki/WSL2-Setup - Copy Paste and Read the Readme currently.

I'm going to close this and put a note in the [FAQ ](https://github.com/oyvindln/vhs-decode/wiki/FAQ)untill we have a Windows distro version released to save any confusion.

closed 2022-10-01 by harrypm