oyvindln/vhs-decode #56: python version of gen_chroma_vid.sh

oyvindln, 2022-08-29:
It would be nicer if we ported these to python, or at least had a python variant that could be used on windows and elsewhere where bash isn't readily available.

harrypm, 2022-08-31:
As mentoned before on the discord, might be worth adding the ability to export media to ld-analyse as an add-on in terms of stream-lined handling adding visual to framing adjustment & chorma decoder settings etc

harrypm, 2023-10-14:
This has now been achieved with [tbc-video-export](https://github.com/JuniorIsAJitterbug/tbc-video-export/releases/tag/v0.0.10) and docs switchover is underway, this has now allowed 100% native windows suite usage for video. (hifi-decode as of present builds)

Tidbit note of [Vrunk11's](https://discord.com/channels/665557267189334046/1076310579817496576/1150890957831688362)  windows bat script [ld-chroma-yc.bat.txt](https://github.com/oyvindln/vhs-decode/files/12906657/ld-chroma-yc.bat.txt).

closed 2023-10-14 by harrypm