oyvindln/vhs-decode #87: vhs-decode not decoding properly since recent Video 8 change

davidvankemenade, 2023-05-09:
After this commit (https://github.com/happycube/ld-decode/commit/0a8eb3449c38abeea2c47adbc984953ebe2547ea) chroma seems to be broken. I previously decoded a Betamax sample which was perfectly fine. Today, I decoded the same capture file and the chroma was off and unstable over time.

Could it be that some NTSC parameters are applied to PAL?

Working well before last commit:
[image]

Chroma off after last commit:
[image]

oyvindln, 2023-05-09:
Yeah I think I might have accidentally skipped the bits that make it not rotation on the chroma signal on PAL betamax (might have also broken u-matic.)

oyvindln, 2023-05-09:
Should be sorted in 524b30e31d2cb5fd3625c72d8ff916458ef76080. Betamax still needs some work in general to look correct though.

davidvankemenade, 2023-05-10:
Many thanks for sorting this out so quickly @oyvindln!

From what I've tried so far, Betamax looks pretty good. What still needs to be optimised?

closed 2023-05-10 by davidvankemenade