oyvindln/vhs-decode #89: PAL-N support?

santiis2010, 2023-05-22:
Hi I was wondering if this has PAL-N support? For Argentina and Uruguay region before I start to setup everything, it says PAL-M that’s only for Brazil.

oyvindln, 2023-05-23:
There isn't explicit support for it at the moment, and don't have any samples of it. The on-tape format is very close to standard PAL though so it might work with just the current PAL settings.

santiis2010, 2023-06-09:
> There isn't explicit support for it at the moment, and don't have any samples of it. The on-tape format is very close to standard PAL though so it might work with just the current PAL settings.

I would be more than fine to send you some digital copies of PAL-N VHS tapes so you can cover all the PAL regions that exist.

Just give an email address i can send to you next week since tomorrow im traveling to the finall of the FIFA World Cup U20 and im not back till monday.

harrypm, 2023-06-10:
@santiis2010 Please put them up on the [community shared drive](https://drive.google.com/drive/u/1/folders/1lzQWdFFfVclEQUDbuwngro0MCusOgPM6?usp=sharing) this is ware we pool and collect sample data currently 9TB of it.

santiis2010, 2023-06-10:
Perfect I will do it on Monday

harrypm, 2024-12-31:
@santiis2010 Did you ever upload any samples for PAL-N?