oyvindln/vhs-decode #91: Would there be any point in implementing support for D-2 video?

vonj, 2023-06-01:
It struck me that even digital tapes could be captured in a similar way as analog tapes. But is there any point in doing so, or do the tape decks capable of reading these D-1, D-2, D-3 etc tapes already come with SDI output, thus a simpler way to get the data into a computer with SDI capture?

https://en.wikipedia.org/wiki/D-2_(video)

harrypm, 2023-06-01:
D2 etc is 4:2:2 Digital Composite (TBC format used for vhs-decode is simaller) so with V210/FFV1 SDI capture workflows anyone can easily run tapes.

DVHS, Like Digital8/MiniDV/HDV/DVCPro/HDCAM/HDCAM-SR etc are all digital formats that fall into the Firewire/SDI capture realm, while its possible to capture the RF via test points it would require new code from the ground up to basically get what you already have from exsiting methods.

The scope of the project is analouge formats from EIAJ to SMPTE open reels formats for example, the list is getting samller each year with now the first of Philps formats "VCR" support just being implimanted.

bradriley, 2023-07-20:
> Or, on a similar note, can D-Theater tapes be "played" in a regular VHS deck?

I would say "not possible", given that D-VHS machines themselves don't use their (S-)VHS heads to play D-VHS tapes.

For what it's worth, the well-respected video expert Disclord said that D-VHS heads used Sendust alloy: https://forum.lddb.com/viewtopic.php?p=38494&sid=83a2b5cb8c2bd4794093b4c4ad7adb97#p38494

harrypm, 2023-10-14:
Notes we have a [Digital Tape Guide](https://github.com/oyvindln/vhs-decode/wiki/Digital-Tape-Guide).

closed 2023-10-14 by harrypm