oyvindln/vhs-decode #92: Betamax PAL: Chroma artefacts in bright scene

davidvankemenade, 2023-06-15:
Betamax PAL seems to work well generally with vhs-decode, but I came across a specific scene with chroma artefacts in the tbc files and the mkv from gen_chroma_vid.

RF capture through DdD: https://drive.google.com/file/d/1MrCZNKYiI4R1SLcS8kQNH37Qh7fZUKRy/view?usp=drive_link
Video showing gen_chroma_vid output: https://youtu.be/BGLGds2iA8Q
Video showing analog video capture: https://youtu.be/g2VQmMFTGH4

Other scenes (e.g. the first and the last bit of the capture) are fine, only the middle song exhibits the chroma artefacts. They seem to coincide with wide shots with a lot of backlight.

Is there anything I can do to avoid these artefacts, or do these scenes show up any known limitations of the current software?

vhs-decode command line:
./vhs-decode -s 3085 -l 150 --no_resample --level_detect_divisor 1 --fallback_vsync --chroma_trap --recheck_phase --pal --threads 1 --tape_format Betamax /home/user/capture.flac /home/user/BetamaxDecode

I've also tried with --cafc, omitting --chroma_trap, with -nocomb and with --clamp, but none of these seem to have an impact on the chroma issue.

Sample frame from ld-analyse:
[image]
Sample frame from video capture:
[image]

atsampson, 2023-06-15:
With the options you've listed above, the TBC struggles all the way through that clip. With just `vhs-decode --no_resample --pal --clamp --tape_format Betamax`, it starts off OK, then the middle shot has some noise overlaid on it (which looks like it's very near the subcarrier frequency - you can see it beating with the chroma burst), then the noise fades out as they fade back to the other camera...

Assuming it's a fault in the original recording, I wonder if we might benefit from a sharper filter in front of ld-chroma-decoder's burst phase detector?

oyvindln, 2023-09-29:
I think this should be mostly fixed by 844ea5d, at least it looks fine when testing here.

davidvankemenade, 2023-10-01:
Hi @oyvindln, many thanks for this fix! I've also tested it and everything seems perfectly fine now. I'll close this issue.

closed 2023-10-01 by davidvankemenade