oyvindln/vhs-decode #100: Merge Python 3.11 numba fixes?

bradriley, 2023-07-20:
Is it possible to just merge Happycube's June 18 ld-decode fix?

https://github.com/happycube/ld-decode/pull/861/commits/2cdcf59c3fed7b8792aef945a676d8453a681ecf

With Python 3.11 currently, vhs-decode (and this branch's ld-decode, ld-cut...) always return the following:

```
/opt/homebrew/lib/python3.11/site-packages/lddecode/utils.py:826: NumbaDeprecationWarning: The 'nopython' keyword argument was not supplied to the 'numba.jit' decorator. The implicit default value for this argument is currently False, but it will be changed to True in Numba 0.59.0. See https://numba.readthedocs.io/en/stable/reference/deprecation.html#deprecation-of-object-mode-fall-back-behaviour-when-using-jit for details.
  @jit(cache=True)
```

(Running MacOS ARM native via Homebrew, if it matters.)

oyvindln, 2023-07-20:
It seems this was committed before the other big changes so I can probably merge these commits a bit easier before fixing the other stuff.

oyvindln, 2023-08-07:
I cherry picked and added this commit so let me know if you're still seeing the warning.

bradriley, 2023-08-18:
This is resolved now; thank you!

(Was on vacation at the time you committed the fix.)

closed 2023-08-18 by bradriley