oyvindln/vhs-decode #109: Sync detection error ntsc-j vha tape

Infrid, 2023-10-05:
There is something that upsets the sync algorithm, I've been discussed about this problem on discord for a while, seems worth to open an issue here.

This Japanese tape has been transferred with DoD by Mighty @harrypm , so I assume at this stage everything was done correctly 😅, at some point there is a very visible sync issue

https://github.com/oyvindln/vhs-decode/assets/3205312/20f50c02-f510-47ad-babc-99ed2ac41c85

command used 

`./decode.py vhs --debug -f 40  --tape_format vhs --system NTSC --recheck_phase --threads 8 debug/ac3-mission00-sample-sync-error.ldf debug/ac3-sample-plain`

Things are a bit better if I append `--level_detect_divisor 6` but still not optimal result

https://github.com/oyvindln/vhs-decode/assets/3205312/2580d6d2-dc29-4541-a6df-c2a618802843

`./decode.py vhs --debug -f 40  --tape_format vhs --system NTSC --recheck_phase --threads 8 --level_detect_divisor 6 debug/ac3-mission00-sample-sync-error.ldf debug/ac3-sample-plain`

You can find the sample .LDF file here https://mega.nz/folder/p4RSAIAA#LzBCNH_NkIac-Ob-CodvHA

I am using the latest version of the decoder from vhs_decode branch

To be fair, @harrypm managed to pull out a stable image eventually using this command

`decode.exe vhs --debug --tape_format vhs --system ntsc --recheck_phase --chroma_trap --threads 8 --high_boost 6`

https://github.com/oyvindln/vhs-decode/assets/3205312/6dcf7dc6-05b4-45cd-9877-75298c28493a

but that version doesn't retain the detail I wanted and it was a compromise to fix another problem on high frequencies. I can give you details on that, but it's not related to this sync error.

thanks!

Infrid, 2023-10-06:
Additional notes from the chat, especially for me, so they won't get lost in discord

> hmm something is screwing with level detection on those frames for whatever reason, if you use 
> `--use_saved_levels`
> it doesn't happen

Infrid, 2023-10-06:
thank you so much, here a before/after comparison for the users out there!

https://github.com/oyvindln/vhs-decode/assets/3205312/3807e21f-fcc8-45de-8691-f6039c67e400

https://github.com/oyvindln/vhs-decode/assets/3205312/85767108-a99a-4327-89c6-0bb1d5a7066f

oyvindln, 2023-10-06:
Something in the signal causes the vsync serration algorithm to interpret something as vertical sync that isn't, and detect the completely wrong levels, and something with the fallback/failsafe logic is failing. I'm not entirely sure what but looking at it I think some of this code needs a bit of rework.

If you just want to decode it for now running with ```----use_saved_levels``` should bypass the problem since it just re-uses the detected levels from the first good field.

harrypm, 2023-10-07:
Note of [Discord Chain](https://discord.com/channels/665557267189334046/665834485975351307/1159468806557683742)

harrypm, 2024-12-31:
Issue no longer present on current code, with newer captures with the ADA4857 amplifyer on the same tape and diff copy of same tape. 

[Ace Combat 3 Electrosphere: Mission 00 - Internet Archive](https://archive.org/details/ace_combat_3_electrosphere_mission_00_fm_rf_vhs_ntsc-j)

closed 2024-12-31 by harrypm

harrypm, 2025-05-24:
Segment issue data has been uploaded to the shared drive [here](https://drive.google.com/drive/u/0/folders/1R5Jrg6_V88D4V4x8cQr9rkNdErixFB6X) as I was doing a re-run and decode and the current code still has this issue might be worth playing with some more @oyvindln