oyvindln/vhs-decode #114: Cannot decode Betamax if compressed

davidvankemenade, 2023-11-04:
**Bug Description**

Compressed Betamax capture cannot be decoded, will yield continuous "Unable to determine start of field - dropping field".

**Steps to Reproduce**

1. Capture 16-bit .raw file using DdD
2. Compress to 16-bit FLAC:
> decode.exe vhs --use_saved_levels --level_detect_divisor 2 --fallback_vsync --chroma_trap --recheck_phase --pal --threads 4 --tape_format Betamax capture.raw outputname

**Expected Behavior**

16-bit FLAC from Betamax would be decoded normally.

**Actual Behavior**

1. Console outputs continous "Unable to determine start of field - dropping field"
2. Log file contains repetition of:
> 2023-11-04 19:19:22,518 - lddecode - DEBUG - VBI EQ serration pulses search failed (using fallback logic)
> 2023-11-04 19:19:22,569 - lddecode - DEBUG - Level detection failed - sync or blank is None
> 2023-11-04 19:19:22,593 - lddecode - DEBUG - No vsync found!
> 2023-11-04 19:19:22,595 - lddecode - ERROR - Unable to determine start of field - dropping field
4. No JSON file is created
5. .tbc and chroma.tbc files are zero bytes

**Environment**

- vhs-decode.exe 0.2.2 (Windows)
- latest commit (Ubuntu) - no version number shown

**Additional Information**

- I can only get 16-bit Betamax captures working by working directly off uncompressed .raw file AND omitting --no_resample
- When using (S)VHS, there is no problem using 16-bit compressed FLAC files as input

davidvankemenade, 2023-11-04:
Apologies, just after submitting this I found that I inadvertently dealt with a compressed Video8 capture instead of the Betamax one. This explains why the syncs couldn't be found.

harrypm, 2023-11-05:
I'll close this then.

closed 2023-11-05 by harrypm