oyvindln/vhs-decode #115: decode.exe 0.2.2 cannot successfully run in hifi mode

davidvankemenade, 2023-11-23:
### Checklist

- [X] I have searched the issues page for any duplicate issues open or closed and confirmed that this bug has not been reported before.
- [X] I have tested the issue with the current build.
- [X] I have attached log files, uploaded sample data, and commands used so that the issue can be easily reproduced by the developers.

### Bug Description

The windows build of decode.exe cannot successfully run in hifi mode.

### Steps to Reproduce

1. Download https://github.com/oyvindln/vhs-decode/releases/download/0.2.2/decode-suite-full.zip
2. Run decode.exe hifi with suitable commands, e.g. `decode.exe hifi -p -f 40 input.raw output.flac`

### Expected Behavior

output.flac is generated

### Actual Behavior

decode.exe returns:
`Traceback (most recent call last):`
`File "decode.py", line 39, in <module>`
`File "decode.py", line 33, in main`
`TypeError: main() takes 0 positional arguments but 1 was given`
`[3768] Failed to execute script 'decode' due to unhandled exception!`

### Environment

- Decode version: 0.2.2
- Operating System: Windows 10
- Hardware Used: AMD Ryzen 9 3900X 12-Core Processor

### Additional Information

vhs mode works fine.

closed 2023-11-25 by oyvindln

davidvankemenade, 2023-11-27:
Thanks for fixing this so quickly @oyvindln!

Would you be happy to add a new Windows release? Or could I build it myself?

oyvindln, 2023-12-03:
I've been aware of the issue for a while but I kinda forgot to make an issue here so kinda forgot about it until now.

Can make a new build now if you want, if not I was hoping to wait a little longer until python 3.12 support in numba is merged which ought to be done sometime very soon which would let me use python 3.12 which may or may not give a slight speed boost.

You can run the latest commits yourself on windows as well of course if you have python (3.11, 3.12 won't work untill numba has merged 3.12 support). The only tricky bit is that you will need to install the exact  microsoft visual studio LTSC build tools matching the version that the python version used was built with to build the cython modules. (For python 3.11 that shoulod be 17.6, you can find it [here](https://learn.microsoft.com/en-us/visualstudio/releases/2022/release-history)) If you have MSVC LTSC installed you use the x64 build tools environment shortcut command line prompt to get into an enviromnent with that available. That lets you either make a python wheel to install or ```python3 setup.py build_ext --inplace``` to build the modules to run directly from the folder.

commit e1cf91a7 (2023-11-25): Put variables in hifi-decode in local scope and fix so it can be ran from decode.py

should fix #115