oyvindln/vhs-decode #122: Cross Platform Support

harrypm, 2024-01-19:
# Operating System Support 

This issue tracks the support level between oprating systems for the decode family of tools and decoders.

## Decoders & Tools - Native Building

- [x] Linux
- [ ] Linux (Arm) (needs testing)
- [x] MacOS (Arm)
- [ ] MacOS (X86) 
- [x] Windows Subsystem For Linux (WSL2)

## Decoders & Tools - Self Contained Binarys

Decoders

- [ ] Linux Appimage
- [x] Windows Binary
- [ ] MacOS App
- [ ] Automated Build Creation With Updates 

LD Tools

- [x] LD-Tools Appimage
- [x] LD-Tools Windows Binary
- [ ] LD-Tools MacOS App
- [ ] Automated Build Creation With Updates 

TBC-Video-Export

- [x] Windows Binary
- [ ] Linux Appimage (work in progress)
- [ ] MacOS App (work in progress)
- [x] Automated Build Creation With Updates 

## Docs 

Command Arguments - 100% Covered (universal)
Post Processing - FFmpeg based & being worked on (self contained with the export tool)
Hardware - MacOS Docs are lacking, and more hardware needs to be vetted, but DdD works today.

oyvindln, 2024-01-19:
I don't think it's really much purpose making an appimage for tbc-video-export. Linux distros are going to have python available anyway so it's easier to just use pypi and pix (or a python virtualenv) to install it.

harrypm, 2024-01-19:
The general thinking is complete self-contained packages for every system type, as we now have zero external dependencies support for Windows, every requirement can be bundled together in a single folder completely offline.

Linux doesn't make sense in a connected environment but for an offline environment and if something breaks and there is no hope of fixing it to the laymam other than just nuking the entire operating system install and starting from fresh, self-contained binaries completely negate the issue, especially if things change on a per-version basis this allows also easy access to older versions when the docs change or a very particular feature changes and I user wants to replicate something.

This would also open up more Linux distro support without maintaining more dependency support...

Also @oyvindln we need documentation for native building on windows clearly laid out otherwise I'm not confident in stating upfront workflow support other than outside of developers, only a handful of people use it natively built.

oyvindln, 2024-01-19:
Yeah I need to write up windows native build instructions

closed 2026-03-19 by harrypm