oyvindln/vhs-decode #126: decode.exe 0.2.4: hifi mode not working

davidvankemenade, 2024-02-02:
### Checklist
- ✅ I have searched the issues page for any duplicate issues open or closed and confirmed that this bug has not been reported before.
- ✅ I have tested the issue with the current build.
- ✅ I have attached log files, uploaded sample data, and commands used so that the issue can be easily reproduced by the developers.

### Bug Description
Latest Windows build doesn't work in hifi mode.

### Steps to Reproduce
1. Download Windows release v0.2.4 (https://github.com/oyvindln/vhs-decode/releases/tag/0.2.4)
2. Execute `decode.exe hifi`

### Expected Behaviour
Output:
`usage: decode.exe [-h] [--system system] [-s start] [--start_fileloc start_fileloc] [-l length] [--overwrite]
                  [-f FREQ] [--cxadc] [--cxadc3] [--10cxadc] [--10cxadc3] [-t threads] [--ct] [--sl sharpness]
                  [--notch notch] [--notch_q notch_q] [-p] [-n] [--pm] [--NTSCJ] [--debug] [--skip_hsync_refine]
                  [--bg] [--preview] [--original] [--noise_reduction NOISE_REDUCTION]
                  [--auto_fine_tune AUTO_FINE_TUNE] [--NR_sidechain_gain NR_SIDE_GAIN] [--h8] [--gnuradio]
                  infile outfile
decode.exe: error: the following arguments are required: infile, outfile`

### Actual Behaviour
Output:
`Traceback (most recent call last): File "decode.py", line 45, in <module> File "decode.py", line 37, in main File "PyInstaller\loader\pyimod02_importers.py", line 419, in exec_module File "vhsdecode\hifi\main.py", line 19, in <module> File "PyInstaller\loader\pyimod02_importers.py", line 419, in exec_module File "vhsdecode\hifi\HiFiDecode.py", line 14, in <module> File "PyInstaller\loader\pyimod02_importers.py", line 419, in exec_module File "pyhht\__init__.py", line 1, in <module> File "PyInstaller\loader\pyimod02_importers.py", line 419, in exec_module File "pyhht\emd.py", line 15, in <module> File "PyInstaller\loader\pyimod02_importers.py", line 419, in exec_module File "pyhht\utils.py", line 15, in <module> ImportError: cannot import name 'angle' from 'scipy' (C:\Users\David\AppData\Local\Temp\_MEI277282\scipy\__init__.pyc) [29828] Failed to execute script 'decode' due to unhandled exception!`

### Environment
Windows 10

### Additional Information
N/A

### Is it related to tbc-video-export?
No

oyvindln, 2024-02-02:
I've made a new build now which should work, and has a bunch of other updates.

davidvankemenade, 2024-02-02:
Thanks @oyvindln for doing this so quickly! hifi seems to work fine now, and I see significant improvements on Video8 and Hi8 - great news.

Closing the hifi issue.

closed 2024-02-02 by davidvankemenade