oyvindln/vhs-decode #137: Add Max Voltage to Recommended Amplifier Board

aldude999, 2024-02-29:
### Bug Description
On the [CX-Cards](https://github.com/oyvindln/vhs-decode/wiki/CX-Cards) wiki page, it mentions that RMS amplifier boards can be powered by 5 or 12 volt rails on the VCR, then in the example product below it, the "With Pot" board only takes 5V max. It should be mentioned that the recommended board does not take more than 5V.

### Additional Information

The amplifier board contains a large amount of magic smoke.

harrypm, 2024-02-29:
`The amplifier board contains a large amount of magic smoke.` 

Lmao aint that the truth yeah thats a typo idk how that got in there without the context of 3-5v range only on the RMS chips.

The new amplifyers however do use 12-30v range and LDOs.

harrypm, 2024-02-29:
[image]

Okay thats sorted.

closed 2024-02-29 by harrypm