oyvindln/vhs-decode #153: cvbs-decode how to decode 54MHz clock sample 10 bit binarys files.

worldsky621-png, 2024-07-09:
### Is your proposal related to a problem?

worldsky621-png, 2024-07-09:
thanks for sharing

harrypm, 2024-07-09:
`-f 54` or `--frequency 54`

[It's covered in the guides](https://github.com/oyvindln/vhs-decode/wiki/RF-Capture-Decoding-Guide)

This is one of the most basic parts of using any decoder setting the frequency and TV System which is pretty much all you set for CVBS decode.

However it's good to note `-A` is still a  required argument to decode most things properly for CVBS-Decode

worldsky621-png, 2024-07-09:
hi，harrypm , thanks for your help. standard raw PCM samples shown below is right?

I use command : python3 cvbs-decode --debug --pal -A -f 54 --system PAL composite-test_54msps_16-bit.u16 composite-test 

[image]

Error Same : Unable to determine start of field - dropping field.

composite-test_54msps_10-bit.u16 sampled data waveform seems ok.
[image]

[image]

worldsky621-png, 2024-07-10:
hi, harrypm. my sample data from cvbs encoder ip( using verilog language + EDA tools) dump data.  

can you share me one standard raw PCM samples files？ I want to see the format and compare data with mine. Thanks very much for your help.

harrypm, 2024-07-10:
Intresting well we have a few samples on the IA page. (I'll add some more later) 

- [WSS PAL Demo Set](https://archive.org/details/wss-wide-screen-signaling) is avalible on the internet archive.

worldsky621-png, 2024-07-10:
hi,harrypm. do you have small data file， similar to composite-test_28msps_8-bit.u8 or so. We just need 8 frames pictures or less.
Thanks a lot for your help.

harrypm, 2024-07-10:
Just take the files in the archive and cut them as you wish they are encoded in FLAC (.cvbs designators)

worldsky621-png, 2024-07-11:
hi，harrypm.  file .cvbs as follows,  format PAL or NTSC ? sample freq ???
[image]

harrypm, 2024-07-11:
The files are 28.6msps the CXADC designator in the name implies stock CX Card usage, the decoders have a --cxadc arguement for this exact crystal rate.

If you actually download and look at the files properly, rather than using internet archives preview you can see exactly what they are in the header data at 1000/1 rate due to limitations of the header of FLAC (but they contain the full sample rate)

Though it is a fair point, those files should be renamed to plane text to prevent outsiders of the projects being confused.

commit 704d08ba (2019-02-28): fix #153
closed 2024-12-01 by harrypm