oyvindln/vhs-decode #154: Decoded hifi audio is noisy at best, or distorted in other cases

davidvankemenade, 2024-07-14:
### Checklist
- ✅ I have searched the issues page for any duplicate issues open or closed and confirmed that this bug has not been reported before.
- ✅ I have tested the issue with the current build.
- ✅ I have attached log files, uploaded sample data, and commands used so that the issue can be easily reproduced by the developers.

### Bug Description
Decoded hifi audio is noisy at best, or distorted in other cases.

### Steps to Reproduce
1. Download sample capture: [Hi8-sitcom-DvK.s16](https://drive.google.com/file/d/1Z5-bJISnIyVb6A9iePyZs58otHGiVBuf/view?usp=drive_link)
2. Decode audio using `decode.exe hifi --pal --frequency 40 --audio_rate 48000 Hi8-sitcom-DvK.s16 Hi8-decoded.flac`
3. Repeat decode with `--8mm` added

### Expected Behaviour
* First section of flac file should be mute, as tape hasn't started yet
* Second section of flac file should be good quality sound without significant noise
* Adding `--8mm` parameter in case of Video8 or Hi8 should improve the decoded sound

### Actual Behaviour
* First section of flac file is pure noise
* Second section of flac file is noisy and somewhat distorted
* When `--8mm` parameter is added, sound is extremely distorted

### Environment
- vhs-decode release 0.2.7
- Operating System: Windows 10

### Additional Information
* It doesn't seem to make much difference whether tape starts after capture has started or not.
* Some sections of large DdD captures sound better than others, it's not yet clear why.

### Is it related to tbc-video-export?
No

harrypm, 2024-07-14:
I believe the primary cause of this issue is the carrier profiles, but most users on the discord have realised it's auto fine tune that was enabled by default a couple updates ago that is causing drastic differences between standard VHS HiFi decoding and 8mm decoding.

I would try with the GUI and disable auto fine tune and try noise reduction on and off and see if that has any difference.

davidvankemenade, 2024-08-31:
Thanks for your response last month @harrypm. Since then, I've tried all kinds of settings but at the moment I'm not getting even close to getting reasonable audio.

Example: https://drive.google.com/file/d/1toeLqyACXoLcp8r_Jns-TY9PPQTr3c57/view?usp=drive_link
This was recorded on a Sony CCD-V6000 camcorder that uses stereo AFM audio.

`decode.exe hifi --threads 4 --pal --frequency 28 --audio_rate 48000 --8mm --audio_mode s DavidVK-Hi8-CCDV6000-90s.flac decoded.flac`

The output is very choppy, which is also visible on the waveform:
[image]

Any ideas about what's going on?

davidvankemenade, 2024-09-28:
I think it's the start of the sample file above that threw hifi-decode off. Still, it now decodes fine using the 0.3.0 Windows build and adding `--auto_fine_tune off` to the command line above.

closed 2024-09-28 by davidvankemenade