oyvindln/vhs-decode #156: hifi-decode fails between after about 7 to 8 minutes of processing time when reading FLAC source file

danifunker, 2024-07-31:
### Checklist

- [✅] I have searched the issues page for any duplicate issues open or closed and confirmed that this bug has not been reported before.
- [✅] I have tested the issue with the current build.
- [✅] I have attached log files, uploaded sample data, and commands used so that the issue can be easily reproduced by the developers.

### Bug Description

When using `hifi-decode` on an 8mm tape on a FLAC compressed file, the application will stop processing. In addition, there is no progress bar that populates.

The same RF file, decompressed is able to process the file.

This specifically was an 8mm tape captured via a regular clock cxadc capture card, but I believe this issue may exist for other captures. I have ran into this bug on multiple captures (all from 8mm sources) and don't believe it is related to my RF captures at all.
 
### Steps to Reproduce

1. Capture the rf signal
2. Convert the rf signal to flac (for reference, I used the following command: `flac --best --sample-rate=28636 --sign=unsigned --channels=1 --endian=little --bps=8 myCapture.u8`
3. `hifi-decode -t 16 --8mm --cxadc --ntsc --auto_fine_tune off --gain .7 --NR_sidechain_gain 80  --audio_mode l --overwrite --audio_rate 192000 myCapture.flac myCapture.wav`

### Expected Behaviour
hifi-decode should complete or at least throw some errors.

### Actual Behaviour

Some time between 7 and 8 minutes the hifi-decode process will hang. Notes: If running the UI, no inputs will be allowed to be clicked. Be aware, prior to the hang the inputs on the UI still work, including pause/start. If running on the console, it just stays there indefinitely until a CTRL+C is pressed.

It's important to note, when processing FLAC files, no progress bar appears, and the wav file is written. Here are some example log outputs. 

16 Thread output:
```
-  Decoding speed: 54355 kFrames/s (0.95x)
- Audio position: 0:07:18.617
- Wall time     : 0:07:42.159
```
1 Thread Output:
```
- Decoding speed: 27304 kFrames/s (0.48x)
- Audio position: 0:03:34.634
- Wall time     : 0:07:30.218
```

### Environment

- Decode version: 0.2.7, as well as the latest commit `8b9a53b592d36cdc1908ce17046ba750942dc90e`
- Operating System: [e.g., Windows 10, macOS Big Sur, Linux Ubuntu] Tried both (wsl2 Linux Ubuntu 24.04) and Linux Ubuntu 24.04.04
- Hardware Used: i7 12700K, 32GB DDR4 RAM, SSD NVME Gen 4 (Samsung 990 PRO), nvidia 3060 graphics card

### Additional Information
Additional reports are coming in on the discord regarding this issue as well

### Is it related to tbc-video-export?
No.

harrypm, 2024-12-03:
This seems to only affect current builds, not the older versions so this is a regression issue.

eshaz, 2025-01-07:
If someone who is able to reproduce the error could follow a `git bisect` and post which commit introduced the issue, that should help narrow down the cause.

I haven't seen this issue on my machines otherwise I'd try this.

davidvankemenade, 2025-01-08:
In my workflow, hifi decoding with long FLAC files (> 90 minutes) completes without any problems. This was already the case with decode.exe 0.2.7, and still now with 0.3.1.

Perhaps this only happens with specific hifi decode parameters? For me, it works with `--8mm --auto_fine_tune off --audio_mode s --threads 4 --pal --frequency 28 --audio_rate 48000`, also without `--8mm`, with `--auto_fine_tune on` and different `frequency` settings.

harrypm, 2025-01-09:
@davidvankemenade Can you also try with the GUI?

davidvankemenade, 2025-01-10:
As mentioned on Discord, I cannot currently get the GUI working on either Windows or Linux. Why would hifi decode behave differently when run through the GUI or through command line?

From Discord: decoding isn't interrupted on Windows using this command: `decode.exe hifi --8mm --auto_fine_tune off --audio_mode s --threads 4 --pal --frequency 28 --audio_rate 48000 tape.video.flac audio.decoded.flac`

It decodes all the way:

```
Initializing ...
PAL Hi8 format selected, Audio mode is s
Starting decode...
Found ffmpeg version N-118193-g5f38c82536-ffmpeg-windows-build-helpers Copyright (c) 2000-2024 the FFmpeg developers

- Decoding speed: 6184 kFrames/s (0.13x)
- Audio position: 0:00:00.585
- Wall time     : 0:00:04.541

(...)

- Decoding speed: 16092 kFrames/s (0.34x)
- Audio position: 1:34:04.170
- Wall time     : 4:40:35.652
Emptying the decode queue ...

Decode finished, seconds elapsed: 16837
Decode finished successfully
```

davidvankemenade, 2025-01-10:
I should add that my original capture was using DdD, 16 bits, then lossy flac compression. Perhaps that makes a difference?

itewreed, 2025-01-26:
I tried a bit of debugging. So whenever a class and function is executed I print out which function in which class it is:
Decoding a raw (u8) file involves, of course, less classes and functions. It seems to only use functions directly defined in the main.py file in the vhsdecode/hifi subfolder
Output
```
Main: log_bias
Bias L -1.31 kHz, R -0.22 kHz (good player/recorder calibration)
Progress [########################################] 99.96%
Main: post_process
Main: prepare Stereo
Main: gain_adjust
Main: gain_adjust
Main: log_bias
Bias L -0.74 kHz, R 0.63 kHz (good player/recorder calibration)
Progress [########################################] 99.98%
Main: post_process
Main: prepare Stereo
Main: gain_adjust
Main: gain_adjust
```

Whereas flac file decoding also involves the use of UnseekableSoundFile and BufferedInputStream Classes.

And this is the output at the moment the decode of a flac file stops:
```
Main: log_bias
Bias L 0.08 kHz, R 1.37 kHz (good player/recorder calibration)
Unseekable soundfile: seek
BufferedInputStream: seek
BufferedInputStream: Tell -> Hack, there is no way to know the current position
Main: post_process
Main: prepare Stereo
Main: gain_adjust
Main: gain_adjust
Unseekable soundfile: read next chunk
BufferedInputStream: read buffer
Buffer has been read...
Data is available
Main: log_bias
Bias L 0.07 kHz, R 1.38 kHz (good player/recorder calibration)
Unseekable soundfile: seek
BufferedInputStream: seek
BufferedInputStream: Tell -> Hack, there is no way to know the current position
Main: post_process
Main: prepare Stereo
Main: gain_adjust
Main: gain_adjust
Unseekable soundfile: read next chunk
BufferedInputStream: read buffer
```
I narrowed it down to the read function in the BufferedInputStream class
```
    def read(self, size=-1):
        print("BufferedInputStream: read buffer")
        data = self.buffer.read(size * 2)
        print("Buffer has been read...")
        if not data:
            print("NO Data!")
            return b""
        else:
            print("Data is available")

        self._pos += len(data)
        return data

```
Where the execution of data = self.buffer.read(size * 2) doesn't work and the decode halts.

Hopefully that helps to figure out the problem.

itewreed, 2025-01-30:
Looked a bit further into the code. Flac files are read by using ffmpeg which pipes it as s16le into hifi-decode. For raw data like raw, u8, u16 etc. the python module called **soundfile** is used. It also supports reading of flac files. So why not natively read flac via **soundfile** module?
see _def as_soundfile_ function
I guess this avoids also buffer problems which seem to occur while reading through ffmpeg.

eshaz, 2025-02-02:
It looks like the update to use Soundfile decodes flac ok, but it's pretty slow.

This turned out to be a better option for me on my machine. Decode with `flac` and pipe the decoded audio into `hifi-decode`. You will need to update the parameters depending on your sample rate and file names.

```sh
flac -d -c --force-raw-format --endian little --sign signed hifi_rf.flac | \
hifi-decode -t 16 -n -f 20MHz - hifi_rf_decoded.flac
```

harrypm, 2025-02-02:
Made it 4:34 sec into a new test capture with the latest commit on a video8 tape that is FLAC compressed...

itewreed, 2025-02-07:
I tested the piping vs. soundfile. The result was, that piping flac into hifi-decode is indeed  faster than letting it run through soundfile. But, this only works properly with 16 Bit flac files. If 8 bit flac files are used, only the way over soundfile works. Piping 8 bit flac files in results in garbage, as hifi-decode expects PCM_16 as input from stdin. So hifi-decode would need to know what it gets feeded in.

eshaz, 2025-02-07:
@itewreed , you can do this as well if `flac` doesn't work. Apparently, there's no option to specify the bit depth for the output, so piping from `flac` will only work with 16 bit files.

```bash
ffmpeg -i hifi_rf.flac -f s16le -acodec pcm_s16le - | \
hifi-decode -t 16 -n -f 20MHz - hifi_rf_decoded.flac
```

closed 2025-11-30 by harrypm