oyvindln/vhs-decode #166: Horizontal sync issue with 8mm

davidvankemenade, 2024-09-27:
### Checklist

- ✅ I have searched the issues page for any duplicate issues open or closed and confirmed that this bug has not been reported before.
- ✅ I have tested the issue with the current build.
- ✅ I have attached log files, uploaded sample data, and commands used so that the issue can be easily reproduced by the developers.

### Bug Description

A specific scene on a Hi8 tape has an unstable hsync:

https://github.com/user-attachments/assets/65358745-59ef-49e8-9d20-c69e27044a52

### Steps to Reproduce

1. Download https://drive.google.com/file/d/1Vlef009iGA5XdWoXEOdaHMnEGDAw0GTh/view?usp=drive_link
2. `decode.exe vhs --ire0_adjust --overwrite  --frequency 40 --level_detect_divisor 1 --fallback_vsync --recheck_phase --pal --threads 4 --tape_format Hi8 DavidVK-Hi8-BigBen.s16 BigBen
3. Observe in ld-analyze or convert to video

### Expected Behaviour

Stable image, constant horizontal sync

### Actual Behaviour

Hsync is unstable in this scene

### Environment

- Decode version: 0.3.0 (Windows build)
- Operating System: Windows 10

### Additional Information

Decoding the rest of the tape, sync seems stable throughout, with the exception of two scenes with buildings against a grey sky.
[BigBen.log](https://github.com/user-attachments/files/17165381/BigBen.log)

### Is it related to tbc-video-export?

No

harrypm, 2024-12-31:
https://github.com/user-attachments/assets/831d437f-a272-467e-81b7-af9b6b2e87dc

Current commit state

davidvankemenade, 2025-01-19:
Big Ben seems to be fine now, with decode.exe 0.3.1 and using `--fallback_vsync`:

https://github.com/user-attachments/assets/b7743e2d-ad70-4bc7-b7a1-fc30792a9c1a

`decode-0.3.1.exe vhs --fallback_vsync --no_resample  --use_saved_levels --ire0_adjust --overwrite --frequency 40 --level_detect_divisor 2 --recheck_phase --pal --threads 4 --tape_format Hi8 DavidVK-Hi8-BigBen.s16 DavidVK-Hi8-BigBen`

Closing this issue.

closed 2025-01-19 by davidvankemenade