oyvindln/vhs-decode #169: Crash in valid_pulses_to_linelocs

philpem, 2024-10-31:
### Checklist

- [✅] I have searched the issues page for any duplicate issues open or closed and confirmed that this bug has not been reported before.
- [✅] I have tested the issue with the current build.
- [✅] I have attached log files, uploaded sample data, and commands used so that the issue can be easily reproduced by the developers.

### Bug Description

When processing video input, since the fix for #167, processing crashes with a traceback. An incomplete TBC file is saved.

### Steps to Reproduce

1. Process recorded CVBS data from CXADC, attached.

[badger.zip](https://github.com/user-attachments/files/17587511/badger.zip)

### Expected Behaviour

Complete TBC file produced.

### Actual Behaviour

CVBS-Decode crashes, producing the following traceback:

```
$ cvbs-decode --overwrite --pal -A --cxadc badger.u8 badger

Missing data at the end of field, possibly dropped samples skipping a little.               
File Frame 11: CAV Pulldown/Telecine Frame                                                  
At field #23, Field phaseID sequence mismatch (8->4) (player may be paused)                 
File Frame 14: CAV Pulldown/Telecine Frame                                      
At field #28, Field phaseID sequence mismatch (8->5) (player may be paused)                 
Exception in thread Thread-70 (decodefield):                                                
Traceback (most recent call last):                                                          
  File "/usr/lib/python3.10/threading.py", line 1016, in _bootstrap_inner                   
    self.run()                     
  File "/usr/lib/python3.10/threading.py", line 953, in run                                 
    self._target(*self._args, **self._kwargs)                                                                                                                                             File "/home/philpem/.local/pipx/venvs/vhs-decode/lib/python3.10/site-packages/lddecode/core.py", line 3674, in decodefield                                                            
    raise e                                                                                 
  File "/home/philpem/.local/pipx/venvs/vhs-decode/lib/python3.10/site-packages/lddecode/core.py", line 3670, in decodefield                                                            
    lpf_wrapper()       
  File "/home/philpem/.local/pipx/venvs/vhs-decode/lib/python3.10/site-packages/lddecode/core.py", line 3055, in process                                                                
    super(FieldPAL, self).process()
  File "/home/philpem/.local/pipx/venvs/vhs-decode/lib/python3.10/site-packages/lddecode/core.py", line 1491, in process                                                                
    self.linelocs1, self.linebad, self.nextfieldoffset = self.compute_linelocs()            
  File "/home/philpem/.local/pipx/venvs/vhs-decode/lib/python3.10/site-packages/cvbsdecode/process.py", line 69, in compute_linelocs                                                    
    linelocs_dict, _ = sync.valid_pulses_to_linelocs(                                       
  File "vhsdecode/sync.pyx", line 146, in vhsdecode.sync.valid_pulses_to_linelocs           
TypeError: an integer is required
saving JSON and exiting 
Exception in thread Thread-71 (decodefield):
Traceback (most recent call last):
  File "/usr/lib/python3.10/threading.py", line 1016, in _bootstrap_inner                   
    self.run()                                                                                                                                                                          
  File "/usr/lib/python3.10/threading.py", line 953, in run                                 
    self._target(*self._args, **self._kwargs)                                               
  File "/home/philpem/.local/pipx/venvs/vhs-decode/lib/python3.10/site-packages/lddecode/core.py", line 3674, in decodefield                                                            
    raise e                                                                                                                                                                             
  File "/home/philpem/.local/pipx/venvs/vhs-decode/lib/python3.10/site-packages/lddecode/core.py", line 3670, in decodefield                                                            
    lpf_wrapper()                                                                           
  File "/home/philpem/.local/pipx/venvs/vhs-decode/lib/python3.10/site-packages/lddecode/core.py", line 3055, in process                                                                
    super(FieldPAL, self).process()                                                                                                                                                       File "/home/philpem/.local/pipx/venvs/vhs-decode/lib/python3.10/site-packages/lddecode/core.py", line 1491, in process                                                                
    self.linelocs1, self.linebad, self.nextfieldoffset = self.compute_linelocs()            
  File "/home/philpem/.local/pipx/venvs/vhs-decode/lib/python3.10/site-packages/cvbsdecode/process.py", line 69, in compute_linelocs                                                    
    linelocs_dict, _ = sync.valid_pulses_to_linelocs(                                                                                                                                   
  File "vhsdecode/sync.pyx", line 146, in vhsdecode.sync.valid_pulses_to_linelocs           
TypeError: an integer is required
```

### Environment

- Decode version: Git rev afd975b1
- Operating System: Linux Mint 21
- Hardware Used: Intel(R) Core(TM) i5-3570K CPU @ 3.40GHz, 8GB RAM

philpem, 2024-10-31:
I chucked a `print()` in here, and the specific issue is that `lastlineloc` is `None`, which causes the crash.

oyvindln, 2024-11-01:
THink 4b3d2eb should fix it

philpem, 2024-11-07:
> THink [4b3d2eb](https://github.com/oyvindln/vhs-decode/commit/4b3d2ebd55ee4ac178b8b37540f7bbf48a7c9a45) should fix it

Confirmed - fixed!

closed 2024-11-07 by harrypm