oyvindln/vhs-decode #173: Using --ct option, while the python samplerate module is installed, gets vhs-decode stuck on Ubuntu 24.04

itewreed, 2024-12-03:
### Checklist

- ✅ I have searched the issues page for any duplicate issues open or closed and confirmed that this bug has not been reported before.
- ✅ I have tested the issue with the current build.
- ✅ I have attached log files, uploaded sample data, and commands used so that the issue can be easily reproduced by the developers.

### Bug Description
When using the --ct option on (k)ubuntu 24.04, while having the samplerate module installed, the decode process immediately gets stuck without reporting any error. Usually by installing the python samplerate module the chromatrap process should be pretty fast.

Doing some debugging in Visual Studio Code I got the following exception from core.py
```
Exception has occurred: AssertionError
<code object calclinelen at 0x77d6813cdde0, file "/home/itereed/git/vhs-decode/lddecode/core.py", line 102> != <code object nb_round at 0x77d6dc4f0430, file "/home/itereed/git/vhs-decode/lddecode/utils.py", line 1019>
KeyError: (1019, '/home/itereed/git/vhs-decode/lddecode/utils.py', <code object nb_round at 0x77d6dc4f0430, file "/home/itereed/git/vhs-decode/lddecode/utils.py", line 1019>)

During handling of the above exception, another exception occurred:

  File "/home/itereed/git/vhs-decode/lddecode/core.py", line 106, in calclinelen
    return int(nb_round(SysParams["line_period"] * mhz * mult))
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/itereed/git/vhs-decode/lddecode/core.py", line 117, in <module>
    SysParams_NTSC["outlinelen"] = calclinelen(SysParams_NTSC, 4, "fsc_mhz")
                                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/itereed/git/vhs-decode/vhsdecode/process.py", line 8, in <module>
    import lddecode.core as ldd
  File "/home/itereed/git/vhs-decode/vhsdecode/main.py", line 18, in <module>
    from vhsdecode.process import VHSDecode
  File "/home/itereed/git/vhs-decode/vhs-decode", line 2, in <module>
    from vhsdecode.main import main
AssertionError: <code object calclinelen at 0x77d6813cdde0, file "/home/itereed/git/vhs-decode/lddecode/core.py", line 102> != <code object nb_round at 0x77d6dc4f0430, file "/home/itereed/git/vhs-decode/lddecode/utils.py", line 1019>
```

### Steps to Reproduce

1. installing vhs-decode according to readme instructions
2. check if samplerate module sees the latest libstdc++ module
3. start a decode, e.g. vhs-decode --frequency 20.000000 --ct --threads 4 --ire0_adjust --recheck_phase --tape_format VHS --system PAL ~/tape_flac/000-099/coll03_tape001_05-20msps.flac ~/2tb-work/tbc/vhshq-test

### Expected Behaviour
Decoding of 5 - 6 frames per second

### Actual Behaviour
No response after starting the process

### Environment

- Decode version: c113eee4b038cee93f3d9df7be5aeaf595b990c6
- Operating System: Kubuntu 24.04, samplerate 0.2.1, libstdc++.so.6 up to GLIBCXX_3.4.33
- Hardware Used: Ryzen 5 7600X, 64GB RAM, SSD NVME

### Additional Information
There are issues with miniconda having not the latest libstdc++ modules on board. So if samplerate is installed and vhs-decode tells samplerate is missing. What it really means in this case is, something like: libstdc++.so.6: version GLIBCXX_3.4.32 not found

### Is it related to tbc-video-export?
No

oyvindln, 2024-12-19:
I'm not able to reproduce on my travel laptop though I'm using OpenSUSE on it. Can't test on Ubuntu until I'm back home in a bit more than a week.

Does clearing the numba cache directory make any difference?
https://numba.readthedocs.io/en/stable/reference/envvars.html#numba-envvars-caching

I don't know if samplerate is the root cause of this issue but I've considered replacing with something else in any case as it requries a workaround to get to build on Fedora and OpenSUSE. Though not sure if the filter effect from the --ct option needs it anyhow, I think a simple band-stop filter could do a similar thing much faster, it's mainly hifi-decode that actually has a use for it.

emarleau, 2025-01-29:
I'm having the same issue on Arch Linux. Clearing the numba cache folder did nothing.