oyvindln/vhs-decode #225: AppImage v0.3.5 can't decode hifi audio, it stalls during the decode

Infrid, 2025-07-25:
Hello, I found out the AppImage version [v0.3.5](https://github.com/oyvindln/vhs-decode/tree/v0.3.5) of the suite, can't really decode hifi audio.

If you download the tool from github, and try to work with a command like this

```vhs-decode-x86_64.AppImage hifi -t 8 --ntsc -f 10 --audio_rate 48000 ../Tape\ 1\ Run\ 4/AC3-Tape-1-Run-4-2025-05-22_18_55_51_01_00-hifi.flac test --overwrite --bg```

After 10 minutes, the software stalls, like if encountered a soft lock. Even the cpu usage gets as low as idle levels. I have to manually terminate the program.

This problem doesn't exist if I compile the tools from the source, in this case everything goes well without any error.

The issue has been also confirmed by @harrypm , I can provide the file to reproduce the bug, just tell me (I am also on discord)

Infrid, 2025-07-25:
I forgot to mention I am using fedora 41

oyvindln, 2025-09-10:
Is it still an issue with the latest version?